mit-han-lab/bevfusion #522: val map=0 error?

mzy991215, 2023-10-11:
Hello, I have recently been doing my related work on your base. Before that, I planned to reproduce your work, but the training results using the provided config were not ideal. The map value was about 0.2, but at the same time, Under one config, I can get ideal evaluation results using the lidar-only-det.pth you provided.

I trained the config configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml on 8*3090, sample per gpu = 4

augment2d:
  gridmask:
    fixed_prob: true
    prob: 0.0
  resize:
  - - 0.38
    - 0.55
  - - 0.48
    - 0.48
  rotate:
  - -5.4
  - 5.4
augment3d:
  rotate:
  - -0.78539816
  - 0.78539816
  scale:
  - 0.9
  - 1.1
  translate: 0.5
checkpoint_config:
  interval: 1
  max_keep_ckpts: 1
cudnn_benchmark: false
data:
  samples_per_gpu: 4
  test:
    ann_file: data/nuscenes/nuscenes_infos_val_radar.pkl
    box_type_3d: LiDAR
    dataset_root: data/nuscenes/
    map_classes:
    - drivable_area
    - ped_crossing
    - walkway
    - stop_line
    - carpark_area
    - divider
    modality:
      use_camera: true
      use_external: false
      use_lidar: true
      use_map: false
      use_radar: false
    object_classes:
    - car
    - truck
    - construction_vehicle
    - bus
    - trailer
    - barrier
    - motorcycle
    - bicycle
    - pedestrian
    - traffic_cone
    pipeline:
    - to_float32: true
      type: LoadMultiViewImageFromFiles
    - coord_type: LIDAR
      load_augmented: null
      load_dim: 5
      reduce_beams: 32
      type: LoadPointsFromFile
      use_dim: 5
    - load_augmented: null
      load_dim: 5
      pad_empty_sweeps: true
[... 1418 lines cut ...]
  - traffic_cone
  type: DefaultFormatBundle3D
- keys:
  - img
  - points
  - radar
  - gt_bboxes_3d
  - gt_labels_3d
  meta_keys:
  - camera_intrinsics
  - camera2ego
  - lidar2ego
  - lidar2camera
  - lidar2image
  - camera2lidar
  - img_aug_matrix
  - lidar_aug_matrix
  type: Collect3D
- keyframe_only: true
  type: GTDepth
use_dim: 5
voxel_size:
- 0.075
- 0.075
- 0.2

mzy991215, 2023-10-11:
> Hello, I have recently been doing my related work on your base. Before that, I planned to reproduce your work, but the training results using the provided config were not ideal. The map value was [...]

{"mode": "val", "epoch": 7, "iter": 753, "lr": 0.00097, "object/car_ap_dist_0.5": 0.2642, "object/car_ap_dist_1.0": 0.3169, "object/car_ap_dist_2.0": 0.3395, "object/car_ap_dist_4.0": 0.3626, "object/car_trans_err": 0.2268, "object/car_scale_err": 0.6048, "object/car_orient_err": 0.6911, "object/car_vel_err": 0.7006, "object/car_attr_err": 0.4029, "object/mATE": 0.4012, "object/mASE": 0.6754, "object/mAOE": 1.3323, "object/mAVE": 0.681, "object/mAAE": 0.3481, "object/truck_ap_dist_0.5": 0.0721, "object/truck_ap_dist_1.0": 0.1297, "object/truck_ap_dist_2.0": 0.1654, "object/truck_ap_dist_4.0": 0.1898, "object/truck_trans_err": 0.4107, "object/truck_scale_err": 0.7505, "object/truck_orient_err": 1.3833, "object/truck_vel_err": 0.522, "object/truck_attr_err": 0.3713, "object/construction_vehicle_ap_dist_0.5": 0.0, "object/construction_vehicle_ap_dist_1.0": 0.0, "object/construction_vehicle_ap_dist_2.0": 0.0, "object/construction_vehicle_ap_dist_4.0": 0.0, "object/construction_vehicle_trans_err": 0.825, "object/construction_vehicle_scale_err": 0.7082, "object/construction_vehicle_orient_err": 1.5372, "object/construction_vehicle_vel_err": 0.1602, "object/construction_vehicle_attr_err": 0.3487, "object/bus_ap_dist_0.5": 0.1389, "object/bus_ap_dist_1.0": 0.2691, "object/bus_ap_dist_2.0": 0.3838, "object/bus_ap_dist_4.0": 0.4262, "object/bus_trans_err": 0.4922, "object/bus_scale_err": 0.8354, "object/bus_orient_err": 1.4944, "object/bus_vel_err": 1.1723, "object/bus_attr_err": 0.4452, "object/trailer_ap_dist_0.5": 0.0016, "object/trailer_ap_dist_1.0": 0.0132, "object/trailer_ap_dist_2.0": 0.0179, "object/trailer_ap_dist_4.0": 0.0292, "object/trailer_trans_err": 0.6123, "object/trailer_scale_err": 0.8539, "object/trailer_orient_err": 1.5, "object/trailer_vel_err": 0.3801, "object/trailer_attr_err": 0.2307, "object/barrier_ap_dist_0.5": 0.0945, "object/barrier_ap_dist_1.0": 0.1456, "object/barrier_ap_dist_2.0": 0.1669, "object/barrier_ap_dist_4.0": 0.1844, "object/barrier_trans_err": 0.3113, "object/barrier_scale_err": 0.6224, "object/barrier_orient_err": 0.6597, "object/barrier_vel_err": NaN, "object/barrier_attr_err": NaN, "object/motorcycle_ap_dist_0.5": 0.0563, "object/motorcycle_ap_dist_1.0": 0.0701, "object/motorcycle_ap_dist_2.0": 0.072, "object/motorcycle_ap_dist_4.0": 0.0729, "object/motorcycle_trans_err": 0.2763, "object/motorcycle_scale_err": 0.7817, "object/motorcycle_orient_err": 1.6117, "object/motorcycle_vel_err": 0.733, "object/motorcycle_attr_err": 0.3897, "object/bicycle_ap_dist_0.5": 0.0, "object/bicycle_ap_dist_1.0": 0.0, "object/bicycle_ap_dist_2.0": 0.0, "object/bicycle_ap_dist_4.0": 0.0, "object/bicycle_trans_err": 0.4067, "object/bicycle_scale_err": 0.7829, "object/bicycle_orient_err": 1.5881, "object/bicycle_vel_err": 0.9705, "object/bicycle_attr_err": 0.1348, "object/pedestrian_ap_dist_0.5": 0.318, "object/pedestrian_ap_dist_1.0": 0.355, "object/pedestrian_ap_dist_2.0": 0.3744, "object/pedestrian_ap_dist_4.0": 0.4045, "object/pedestrian_trans_err": 0.2625, "object/pedestrian_scale_err": 0.4387, "object/pedestrian_orient_err": 1.5251, "object/pedestrian_vel_err": 0.8091, "object/pedestrian_attr_err": 0.4612, "object/traffic_cone_ap_dist_0.5": 0.0803, "object/traffic_cone_ap_dist_1.0": 0.0837, "object/traffic_cone_ap_dist_2.0": 0.0922, "object/traffic_cone_ap_dist_4.0": 0.1135, "object/traffic_cone_trans_err": 0.1884, "object/traffic_cone_scale_err": 0.3754, "object/traffic_cone_orient_err": NaN, "object/traffic_cone_vel_err": NaN, "object/traffic_cone_attr_err": NaN, "object/nds": 0.26199, "object/map": 0.14511}

I have trained 7 epochs，map only 0.14511？

mzy991215, 2023-10-11:
> Hello, I have recently been doing my related work on your base. Before that, I planned to reproduce your work, but the training results using the provided config were not ideal. The map value was [...]

zyh321, 2024-05-21:
> Hello, I have recently been doing my related work on your base. Before that, I planned to reproduce your work, but the training results using the provided config were not ideal. The map value was [...]

@mzy991215 Hi, I am also repruducing the lidar-only result, but the mAP is low. Do you resolve this problem？