mit-han-lab/bevfusion #529: How to debug train.py through vscode?

lrbob, 2023-10-17:
如何使用"./vscode/launch.json"
[image]
[image]
 debug train.py代码？

{
"version": "0.2.0",
"configurations": [

      {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "console": "integratedTerminal",
            "args": [
                "--np","2",
                "tools/train.py",
            ],
            "env": {"CUDA_VISIBLE_DEVICES":"0,1"},

             "program": "/opt/conda/lib/python3.8/site-packages/torch/distributed/run.py",
      }
    ]
}
这样设置以后运行debug还是提示KeyError: 'MASTER_HOST'
[image]
[image]

chfyyds, 2023-10-20:
I also encountered the same problem in pycharm. Did you solve it?

lrbob, 2023-10-20:
> I also encountered the same problem in pycharm. Did you solve it?

oh yes, run command on the current bevfusion env terminal:
"export MASTER_HOST=your ipv4 address : a port number not used"

closed 2024-05-04 by zhijian-liu