mit-han-lab/bevfusion #531: the MAP is very low,

mzy991215, 2023-10-17:
I use your config(lidar-only), and I got the 0 MAP. But  when i use the .pth you provided(lidar-only-det.pth) to eval, I can reporduce the standard result. 
and then I thought there may be a bug in dataset pipeline, so I banned the ObjectPaste，but I still got the wrong mAP about 0.249

Could you help me, thank you!

mAP: 0.0459                              
mATE: 0.5196
mASE: 0.4398
mAOE: 0.9982
mAVE: 1.0748
mAAE: 0.4462
NDS: 0.1826
Eval time: 101.5s
Per-class results:
Object Class            AP      ATE     ASE     AOE     AVE     AAE   
car                     0.068   0.356   0.462   0.695   1.220   0.428 
truck                   0.015   0.593   0.376   0.585   0.748   0.401 
bus                     0.061   0.685   0.301   0.893   1.805   0.614 
trailer                 0.000   0.662   0.328   1.251   0.473   0.346 
construction_vehicle    0.000   0.910   0.508   1.550   0.153   0.352 
pedestrian              0.226   0.357   0.483   1.144   0.996   0.799 
motorcycle              0.000   0.415   0.463   1.189   2.566   0.461 
bicycle                 0.000   0.418   0.556   1.425   0.638   0.168 
traffic_cone            0.013   0.296   0.505   nan     nan     nan   
barrier                 0.075   0.505   0.416   0.253   nan     nan

Li-Whasaka, 2024-01-23:
Have you found problem?

SivenCapo, 2024-04-08:
Have you found problem? I encountered the same issue for 1 month.
Could we have a Communication?

closed 2024-07-31 by zhijian-liu