mit-han-lab/bevfusion #538: error for segmentation  models in training and test

heli223, 2023-10-26:
**when i run the commands for segmentation with any model (camera-only, lidar-only, bevfusion), it will occur errors.

1. For training, its error is about** 
File "/home/bevfusion-main/mmdet3d/datasets/pipelines/transforms_3d.py", line 218, in __call__
    data["radar"].rotate(-theta)
AttributeError: 'dict' object has no attribute 'rotate'

**2. For test, its error is about**
File "/home/bevfusion-main/mmdet3d/datasets/pipelines/loading.py", line 294, in __call__
    location = data["location"]
KeyError: 'location'

can you help me with this? thanks!

Yunge6666, 2023-11-06:
I have the same 'location' error. Have you solved this problem?

heli223, 2023-11-07:
> I have the same 'location' error. Have you solved this problem?
I used an environment build by my friend, and it is okay now. Maybe I think it is because the versions of different packages. I don't know which packages are the reasons since the most important packages are the same.

closed 2023-11-20 by heli223

971022jing, 2024-01-04:
> **when i run the commands for segmentation with any model (camera-only, lidar-only, bevfusion), it will occur errors.** **1. For training, its error is about** File [...]

hello，have you solved the problems?

heli223, 2024-01-04:
> hello，have you solved the problems?

No, I did not solve it. I used an old version of the code at https://gitee.com/linClubs/bevfusion. Maybe you can have a look.

EpicGilgamesh, 2024-01-05:
@heli223 you had just downloaded this older version and everything is working fine?
I'm fighting rn for over a month with BEVFusion, because I need it for my engineer thesis and after fixing one thing there is always another...
Is this older version fine?
Also if you could list used packages pls

EpicGilgamesh, 2024-01-05:
@heli223  you had just downloaded this older version and everything is working fine?
I'm fighting rn for over a month with BEVFusion, because I need it for my engineer thesis and after fixing one thing there is always another...
Is this older version fine? I see that it has newer commits that the one here, you downloaded whole new repo or just multiple files?
Also if you could list used packages pls

EpicGilgamesh, 2024-01-05:
@heli223 you had just downloaded this older version and everything is working fine?
I'm fighting rn for over a month with BEVFusion, because I need it for my engineer thesis and after fixing one thing there is always another...
I see that it has newer commits that the one here, you downloaded whole new repo or just multiple files?
Also if you could list used packages pls

heli223, 2024-01-08:
> @heli223 you had just downloaded this older version and everything is working fine? I'm fighting rn for over a month with BEVFusion, because I need it for my engineer thesis and after fixing one [...]

I download the whole repo at https://gitee.com/linClubs/bevfusion and use it according to the following link: https://zhuanlan.zhihu.com/p/649200166

CaveSpiderLZJ, 2024-01-19:
To solve the first error, you can just comment these lines in `mmdet3d -> datasets -> pipelines -> transforms_3d.py -> GlobalRotScaleTrans -> __call__` like this:
```
if "points" in data:
    data["points"].rotate(-theta)
    data["points"].translate(translation)
    data["points"].scale(scale)

# if "radar" in data:
#     data["radar"].rotate(-theta)
#     data["radar"].translate(translation)
#     data["radar"].scale(scale)

gt_boxes = data["gt_bboxes_3d"]
rotation = rotation @ gt_boxes.rotate(theta).numpy()
gt_boxes.translate(translation)
gt_boxes.scale(scale)
data["gt_bboxes_3d"] = gt_boxes
```