mit-han-lab/bevfusion #540: ValueError: matrix contains invalid numeric entries

1653042420, 2023-10-27:
Hi! Thanks for sharing your excellent work! when I trained the Lidar branch
I only have one RTX 4080，and my batch size is 4. I encountered this issue when using the previously released code to train the LiDAR branch.Now,I know that it was caused by an incorrect learning rate. I want to konw if I use the latest released code, do I still need to adjust the learning rate based on the total batch size？

971022jing, 2024-01-17:
> Hi! Thanks for sharing your excellent work! when I trained the Lidar branch I only have one RTX 4080，and my batch size is 4. I encountered this issue when using the previously released code to train [...]

I have the same problem. Do you have any new progress?

nanqiang-zhangzhaoxu, 2024-02-01:
I have the same problem. Do you have any new progress?

wyf0414, 2024-04-02:
I have the same problem, too. And when I increase the `max_epoch`, the corresponding` lr` needs to be smaller. I have to adjust the `lr` again and again.

gerardmartin2, 2024-05-26:
Hi, I have adjusted also the learning rate but in the 5th epoch it starts to slow down a lot. If you have made any modification in the lr schedule, can you show it? As my batch_size if 3 (approx 1/10 of the original) I have changed lr (from optimizer and min_lr_ratio) to 1/10 of original. Before this change my training was stucked at epoch2 and now it reaches epoch5, but as said, it starts to go too slow.

Thanks in advance

zyqww, 2024-07-12:
@gerardmartin2 Hello, has the above method been successfully reproduced?Looking forward to your reply!

gerardmartin2, 2024-07-29:
Hello @zyqww, sorry for the delay. I had something like 1-2% peformance degradation  with respect to the validation results (training only 4 epochs since, as mentioned, the training did not converge in the 5th). I am not sure that this lack of convergence is completely due to these modifications since I have modified some parts of the mmcv and mmdet libraries too. So maybe you can reach 6-8 epochs with these training modifications and get better results than me.

closed 2024-07-31 by zhijian-liu