mit-han-lab/bevfusion #542: evaluation Lidar-only works, Camera-only dones

Terencedu, 2023-10-30:


Terencedu, 2023-10-30:
Hi,

My GPU is one 4060(laptop, 8GB). Cuda version: 11.8

Environment:
mmcv-full = 1.4.0
mmdet = 2.20.0
mmdet3d = 0.0.0
nuscenes-devkit = 1.1.10
torch = 1.9.1+cu111

When I use "torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml pretrained/lidar-only-det.pth --eval bbox" to evaluate the lidar-only model, it works:
[image]

However, when I use "torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml pretrained/camera-only-det.pth --eval bbox" to evaluate the camera-only model, it fails:
[image]

Is this an environment version issue or OOM issue?

Best wishes!

Terencedu, 2023-10-31:
It was an environmental problem, and now it is solved. Thanks.

closed 2023-10-31 by Terencedu

BeMuCa, 2023-12-22:
@Terencedu  How did you solve the Issue? I have the exact opposite, Image worked, Lidar is failing with the same error handle