mit-han-lab/bevfusion #543: RuntimeError: sigmoid_focal_loss_forward_impl: implementation for device cuda:0 not found

konyul, 2023-10-30:
I followed the instruction on README.md, but got above error.

konyul, 2023-11-01:
The error was generated because the MMCV CUDA Compiler was not available.
MMCV CUDA Compiler is not available in cu113 and torch1.10.0
Therefore creating docker images with this docker file will solve the problem

change the dockerfile with this

FROM nvidia/cuda:11.0.3-cudnn8-devel-ubuntu18.04

RUN apt-get update && apt-get install wget -yq
RUN apt-get install build-essential g++ gcc -y
ENV DEBIAN_FRONTEND noninteractive
RUN apt-get install libgl1-mesa-glx libglib2.0-0 -y
RUN apt-get install openmpi-bin openmpi-common libopenmpi-dev libgtk2.0-dev git -y

# Install miniconda
ENV CONDA_DIR /opt/conda
RUN wget --quiet https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O ~/miniconda.sh && \
     /bin/bash ~/miniconda.sh -b -p /opt/conda
# Put conda in path so we can use conda activate
ENV PATH=$CONDA_DIR/bin:$PATH
RUN conda install python=3.8
RUN conda install pytorch==1.7.0 torchvision==0.8.0 cudatoolkit=11.0 -c pytorch
RUN pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu110/torch1.7/index.html
RUN pip install mmcv==1.4.0 mmdet==2.20.0
RUN pip install Pillow==8.4.0
RUN pip install tqdm
RUN pip install torchpack
RUN pip install nuscenes-devkit
RUN pip install mpi4py==3.0.3
RUN pip install numba==0.48.0

and then command cd docker && docker build . -t bevfusion

closed 2023-11-01 by konyul

viborgen, 2023-11-15:
If it helps anyone I solved this by downloading mmcv from source and then installing it via the following command

MMCV_WITH_OPS=1 FORCE_CUDA=1 pip install -v -e .