mit-han-lab/bevfusion #546: Kitti training exception

zzwforver, 2023-11-06:
Hello, I used BEVfusion to train kitti and changed it to a single frame. After the training, the loss and score were very low, which can ensure that the matrix and its transformation are correct. The loaded annotations and data sources sent to the network are correct, but this problem still exists. I hope to exchange and guide you

HydrogenWasser, 2023-11-17:
have u solved this problem?

i've tried it on kitti too and mAP is very low even if on training set, i got no idea why this happens.

Pedestrian AP40@0.50, 0.50, 0.50:
bbox AP40:0.0005, 0.0003, 0.0007
bev  AP40:0.0000, 0.0000, 0.0000
3d   AP40:0.0000, 0.0000, 0.0000
aos  AP40:0.00, 0.00, 0.00
Pedestrian AP40@0.50, 0.25, 0.25:
bbox AP40:0.0005, 0.0003, 0.0007
bev  AP40:0.0000, 0.0000, 0.0000
3d   AP40:0.0000, 0.0000, 0.0000
aos  AP40:0.00, 0.00, 0.00
Cyclist AP40@0.50, 0.50, 0.50:
bbox AP40:0.0016, 0.0045, 0.0073
bev  AP40:0.0000, 0.0000, 0.0000
3d   AP40:0.0000, 0.0000, 0.0000
aos  AP40:0.00, 0.00, 0.00
Cyclist AP40@0.50, 0.25, 0.25:
bbox AP40:0.0016, 0.0045, 0.0073
bev  AP40:0.0000, 0.0000, 0.0000
3d   AP40:0.0000, 0.0000, 0.0000
aos  AP40:0.00, 0.00, 0.00
Car AP40@0.70, 0.70, 0.70:
bbox AP40:0.0160, 0.0207, 0.0262
bev  AP40:0.0006, 0.0004, 0.0004
3d   AP40:0.0002, 0.0001, 0.0001
aos  AP40:0.01, 0.01, 0.01
Car AP40@0.70, 0.50, 0.50:
bbox AP40:0.0160, 0.0207, 0.0262
bev  AP40:0.0068, 0.0094, 0.0107
3d   AP40:0.0017, 0.0015, 0.0018
aos  AP40:0.01, 0.01, 0.01

Overall AP40@easy, moderate, hard:
bbox AP40:0.0061, 0.0085, 0.0114
bev  AP40:0.0002, 0.0001, 0.0001
3d   AP40:0.0001, 0.0000, 0.0000
aos  AP40:0.00, 0.00, 0.01

liuyuansheng624, 2023-11-18:
> have u solved this problem?
>
> i've tried it on kitti too and mAP is very low even if on training set, i got no idea why this happens.
>
> Pedestrian AP40@0.50, 0.50, 0.50: bbox AP40:0.0005, 0.0003, [...]

你好，请问可以请教一下您是如何使用KITTI进行训练的吗

HydrogenWasser, 2023-11-20:
> 你好，请问可以请教一下您是如何使用KITTI进行训练的吗

重写图片读取的类和运行所需的config文件

liuyuansheng624, 2023-11-20:
> 重写图片读取的类和运行所需的config文件

加您个联系方式可以么

HydrogenWasser, 2023-11-20:
可以，你可以把你想使用的联系方式发到这个邮箱 cloud_hua@126.com

HydrogenWasser, 2023-11-20:
> 加您个联系方式可以么

可以，你可以把你想使用的联系方式发到这个邮箱 [cloud_hua@126.com](mailto:cloud_hua@126.com)

DGCHAO, 2023-11-24:
> > > 你解决这个问题了吗？
> > > 我也在 kitti 上尝试过，即使在训练集上，mAP 也非常低，我不知道为什么会发生这种情况。
> > > 行人 AP40@0.50、0.50、0.50： bbox AP40:0.0005、0.0003、0.0007 bev AP40:0.0000、0.0000、0.0000 3d AP40:0.0000、0.0000、0.0000 aos AP40:0.00、 0.00、0.00 行人 AP40@0.50、0.25、0.25： bbox AP40:0.0005, 0.0003, 0.0007 bev AP40:0.0000, 0.0000, 0.0000 3d AP40:0.0000, 0.0000, 0.0000 aos AP40:0.00, 0.00, 0.00 骑行者 AP40@0.50, 0. 50、0.50：bbox AP40：0.0016、0.0045、0.0073 bev AP40:0.0000, 0.0000, 0.0000 3d AP40:0.0000, 0.0000, 0.0000 aos AP40:0.00, 0.00, 0.00 骑行者 AP40@0.50, 0.25, 0.25: bbox AP40:0.0016, 0.0045, 0.0073 bev AP40：0.0000、0.0000、0.0000 3d AP40 :0.0000, 0.0000, 0.0000 aos AP40:0.00, 0.00, 0.00 汽车 AP40@0.70, 0.70, 0.70: bbox AP40:0.0160, 0.0207, 0.0262 bev AP40:0.0006, 0.0004, 0.00 04 3d AP40：0.0002、0.0001、0.0001 aos AP40： 0.01, 0.01, 0.01 汽车 AP40@0.70, 0.50, 0.50: bbox AP40:0.0160, 0.0207, 0.0262 bev AP40:0.0068, 0.0094, 0.0107 3d AP40:0.0017, 0.0015, 0.00 18 aos AP40:0.01, 0.01, 0.01 总体 AP40@ 
> > > easy ，中等，硬：bbox AP40：0.0061、0.0085、0.0114 bev AP40：0.0002、0.0001、0.0001 3d AP40：0.0001、0.0000、0.0000 aos AP40：0.00、0.00、0.01
> > 
> > 
> > 您好，请问可以教一下您是如何使用 KITTI 进行训练的吗？
> 
> 重写图片读取的类和运行所需的config文件

你好，我也运到相同问题，可以加个联系方式请教一下吗？

HydrogenWasser, 2023-11-24:
> 你好，我也运到相同问题，可以加个联系方式请教一下吗？

请参考我之前的回复，把您的联系方式发到我的邮箱

closed 2024-05-04 by zhijian-liu