mit-han-lab/bevfusion #547: Pretrained Swin-T not available anymore

IdrissGo, 2023-11-06:
I used the command from the script provided to get the weights for Swin-T 
`wget https://bevfusion.mit.edu/files/pretrained_updated/swint-nuimages-pretrained.pth
`
However I get a 404 not found error : 

> Resolving bevfusion.mit.edu (bevfusion.mit.edu)... 18.25.16.171
> Connecting to bevfusion.mit.edu (bevfusion.mit.edu)|18.25.16.171|:443... connected.
> HTTP request sent, awaiting response... 301 Moved Permanently
> Location: http://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth [following]
> --2023-11-06 10:55:34--  http://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth
> Resolving hanlab.mit.edu (hanlab.mit.edu)... 52.17.119.105, 63.35.51.142, 34.249.200.254
> Connecting to hanlab.mit.edu (hanlab.mit.edu)|52.17.119.105|:80... connected.
> HTTP request sent, awaiting response... 301 Moved Permanently
> Location: https://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth [following]
> --2023-11-06 10:55:34--  https://hanlab.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth
> Connecting to hanlab.mit.edu (hanlab.mit.edu)|52.17.119.105|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2023-11-06 10:55:36 ERROR 404: Not Found.

Can you provide a new link to download the pretrained Swin-T ? 

Best regards,

GaoPeng97, 2023-11-08:
try :
wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth

zxsgem, 2023-11-09:
> try : wget http://hanlab18.mit.edu/projects/bevfusion/files/pretrained_updated/swint-nuimages-pretrained.pth

I have solved this problem，thanks for your help ！

closed 2024-05-04 by zhijian-liu