mit-han-lab/bevfusion #551: Training with torchpack is unresponsive

1180100407, 2023-11-12:
I followed the instructions in the readme but there is no output, the problem seems to be with the torchpack but I don't know how to fix it, using "torchpack dist-run -np 1" doesn't produce any output either, and I've waited for at least 10 minutes.

alexzhuuuu, 2023-12-29:
Same problem. What's the reason? I tried to run it without torchpack, and just python the test.py file.

alexzhuuuu, 2023-12-29:
> I followed the instructions in the readme but there is no output, the problem seems to be with the torchpack but I don't know how to fix it, using "torchpack dist-run -np 1" doesn't produce any output either, and I've waited for at least 10 minutes.
Same problem. What's the reason? I tried to run it without torchpack, and just python the test.py file.

Lawrence-Cheung, 2024-04-23:
I met another problem like below:

Traceback (most recent call last):
  File "C:\Users\18369\.conda\envs\bevfusion\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\18369\.conda\envs\bevfusion\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\18369\.conda\envs\bevfusion\Scripts\torchpack.exe\__main__.py", line 7, in <module>
  File "C:\Users\18369\.conda\envs\bevfusion\lib\site-packages\torchpack\launch\main.py", line 17, in main
    drunner.main()
  File "C:\Users\18369\.conda\envs\bevfusion\lib\site-packages\torchpack\launch\launchers\drunner.py", line 104, in main
    os.execve('/bin/sh', ['/bin/sh', '-c', command], environ)
FileNotFoundError: [Errno 2] No such file or directory: '/bin/sh'

who can help?is this problem result in torchpack?

closed 2024-07-31 by zhijian-liu