mit-han-lab/bevfusion #552: nuscenes-mini Performance on TestSet Lower than the official results

wstzx, 2023-11-14:
> Hello author, thank you so much for open sourcing your code. I tried to run your latest code using nuscenes-mini on a single GPU on 3090 and A5000, and it showed: 7.RuntimeError: Given groups=1, weight of size [8, 1, 1, 1], expected input[12, 6, 256, 704] to have 1 channels, but got 6 channels instead , so I setmmdet3d/models/vtransforms/base line37 and line38: False, but the results of the test MAP = 58, and after I used the two other old version to test, the result is only 58, far lower than the final result, is the reason I used one GPU?Here are the results of my single GPU tests in nuscenes-mini: [image]
This is my run command:
torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox

_Originally posted by @wstzx in https://github.com/mit-han-lab/bevfusion/issues/478#issuecomment-1808015449_

ZoeSnow, 2023-11-26:
The same problem. Do you have any new progress?

hankeceli, 2023-12-17:
Hello @wstzx @ZoeSnow @qpc001 ,

I've encountered an issue right at the onset of the dataset preparation using the nuScenes `v1.0-mini`.

I would appreciate it if you could take a moment to review the details in my open [issue](https://github.com/mit-han-lab/bevfusion/issues/569#issue-2042005347). Your insights and assistance would be highly valuable.

Thank you!

zhijian-liu, 2024-05-04:
We haven't used the mini dataset in our experiments. Have you tried the full dataset to see if the same issue still persists?

closed 2024-07-31 by zhijian-liu