mit-han-lab/bevfusion #556: error for test seg task

Excutioner, 2023-11-21:
During run `torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`
I had the same problem as #461at first, but I added the following code after line 185 in `nusences_converter.py`
` file location = nusc.get(
            "log", nusc.get("scene", sample["scene_token"])["log_token"]
        )["location"]`
And added location to the info
`'location': location, `
[image]
Rerun nusecenes_converter to regenerate the PKL file, the following error was encountered
AttributeError: 'dict' object has no attribute 'in_range_bev'
[image]

Yunge6666, 2023-12-08:
Have you solved this problem? I have the same problem...

Qizhi697, 2023-12-29:
Have you solved this problem? I have the same problem...

Qizhi697, 2023-12-30:
I have also got this problem during run `torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`. 
I don't know how to evaluate the pretrained model for the segmentation variant of BEVFusion.

closed 2024-07-31 by zhijian-liu