mit-han-lab/bevfusion #560: ModuleNotFoundError: No module named 'torchpack.utils.tqdm'

Simeon340703, 2023-11-28:
While attempting to run BevFusion (available at bevFusion GitHub(https://github.com/mit-han-lab/bevfusion)) for visualization purposes, I encountered a ModuleNotFoundError. Despite trying different versions of torchpack, such as 0.3.1 and 0.3.0, the issue remained unresolved.  Which version of Torchpack did you use? The training and evaluation  are working. I run the visualization code  using "torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth  --out-dir /pretrained" on a single GPU. 

Here is the complete error message:
Traceback (most recent call last):
  File "tools/visualize.py", line 13, in <module>
    from torchpack.utils.tqdm import tqdm
ModuleNotFoundError: No module named 'torchpack.utils.tqdm'
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[32545,1],0]
  Exit code:    1

Simeon340703, 2023-11-29:
I  have solved it by installing it from the source. The 'pip install torchpack' did not work for me and there was no alternative way to install it.

closed 2023-11-29 by Simeon340703

shupinghu, 2024-05-15:
> I have solved it by installing it from the source. The 'pip install torchpack' did not work for me and there was no alternative way to install it.

Hello, I met the same problem with you. And I also tried to use version 0.3.1 and 0.3.0, install from the source (Downloaded from https://pypi.org/project/torchpack/ and use "python setup.py install" to install it. But this error message still occured. 

Could you please provide your method in detail so that I can solve this problem as well?

Thank you very much!