mit-han-lab/bevfusion #564: Wrong depth map

Jayden9912, 2023-12-05:
Hi. I retrained the model and try to visualize the depth map predicted and find that it doesn't make sense. However, the performance of the model is about the same as recorded in the paper.

[image]
[image]

I am using viridis colour scale.
[image]

liyih, 2023-12-10:
Maybe you should close all the data augmentation and visualize the depth map again. If you don't close them the depth map corresponds to the augmented picture (scale, flip, rot, mask...)

Jayden9912, 2023-12-10:
Thanks for the suggestion. Would give it a try later.

In the meantime, did you get a reasonable depth map on your side? May I have a look at it?

liyih, 2023-12-10:
I only segment the object in the picture and project them into 3d space by depth. And I find the projected visual points overlap well with the original points cloud. In some way, it may prove the depth is right. The 2d pictures and 3d points may not in the same frame as I don't have enough time to do correspondence.
[image]
[image]

[image]
[image]

Jayden9912, 2023-12-12:
Hi, thanks for the visualization. May I know how you got the depth value? Are u using LSS transform for depth prediction as well?

liyih, 2023-12-12:
I'm sorry that I misunderstand your meaning at begin. In fact, if you directly use the depth prediction by depth net it make no sense. Because in bevfusion the output of depth net is not be supervised. In the BaseDepthTransform of original code, it gives the right way to use lidar to get the depth of pixels (you could use this depth map to do visualization). If you want to make the output of depth net have right depth prediction, you should give a supervision like what bevdepth do. Please scan for bevdepth, which may give you some help.
Best

Jayden9912, 2023-12-12:
I see, thanks for the suggestion.

I am quite confused because although the depth predicted by lss transform is not logical but it works with 3d object detection...

Any clue on this?

liyih, 2023-12-12:
please look at part 3.2 in bevdepth, you may get some inspiration.

dingmiaomiao, 2023-12-14:
I added AwareBEVDepth and the corresponding dense depth loss during training, but I found that the loss decreased by less than a small value, and there was no improvement in detection accuracy. Has anyone tried to add this loss?

Jayden9912, 2023-12-14:
I remember in one of the issue the author mentioned the same observation. I am quite curious how accurate is the depth map trained using AwareBEVDepth

closed 2024-05-04 by zhijian-liu