mit-han-lab/bevfusion #566: training strategy

liyih, 2023-12-07:
Hi, authors!
    Thanks for your great work. By the way I change some codes of your work and obverse that the model doesn't converge enough in 6 epochs. So, I want to increase the training time (from 6 epochs to 10 or 12 epochs). I want to know what I should change for the Param scheduler (including LinearLR, CosineAnnealingLR, CosineAnnealingMomentum, CosineAnnealingMomentum). 
Best wish.

186big, 2024-01-05:
Hello，I meet a question at #582 ,I want to ask   what is the result of model convergence after training 6 epochs with the author's original code, and can it achieve the effect in the author's paper?
Thank you ,
Best wish.

liyih, 2024-01-05:
我复现源码的效果和作者的差不多NDS，mAP +- 0.2。我的不收敛发生在我在原代码中加了模块后，但是效果依然正常，只是我觉得第六个epoch效果增长并没有完全停止。问题582中的效果很奇怪，我觉得你可以试着检测一下你生成的data和pickcle是否正确，还有检查一下在load预训练参数的时候有没有出现通道不匹配的情况。你也可以把训练过程中loss的变化发一下，从中可以看出一点端倪。

Li-Whasaka, 2024-01-23:
> 我复现源码的效果和作者的差不多NDS，mAP +- 0.2。我的不收敛发生在我在原代码中加了模块后，但是效果依然正常，只是我觉得第六个epoch效果增长并没有完全停止。问题582中的效果很奇怪，我觉得你可以试着检测一下你生成的data和pickcle是否正确，还有检查一下在load预训练参数的时候有没有出现通道不匹配的情况。你也可以把训练过程中loss的变化发一下，从中可以看出一点端倪。

Could you please share your train log file with me? What is your loss when you achieve the best performance？

Li-Whasaka, 2024-01-23:
OK，you can  attach file in gihub

Li-Whasaka, 2024-01-23:
> > > 我复现源码的效果和作者的差不多NDS，mAP +- 0.2。我的不收敛发生在我在原代码中加了模块后，但是效果依然正常，只是我觉得第六个epoch效果增长并没有完全停止。问题582中的效果很奇怪，我觉得你可以试着检测一下你生成的data和pickcle是否正确，还有检查一下在load预训练参数的时候有没有出现通道不匹配的情况。你也可以把训练过程中loss的变化发一下，从中可以看出一点端倪。
> > 
> > 
> > Could you please share your train log file with me? What is your loss when you achieve the best performance？
> 
> Ok, thank you , I’ll send you later,email?

OK，you can attach file in github

186big, 2024-01-23:
> OK，you can attach file in github

[Uploading 20231225_095320.log.json…]()

186big, 2024-01-23:
> Reference in new

[20231225_095320.log.json](https://github.com/mit-han-lab/bevfusion/files/14018643/20231225_095320.log.json)

186big, 2024-01-23:
> OK，you can attach file in github

[20231225_095320.log.json](https://github.com/mit-han-lab/bevfusion/files/14018645/20231225_095320.log.json)

closed 2024-05-04 by zhijian-liu