mit-han-lab/bevfusion #568: Custom Dataset Implementation

GoodarzMehr, 2023-12-13:
Hi, thanks for the great work!

For training [BEV segmentation with a custom dataset](https://github.com/GoodarzMehr/bevfusion/tree/develop) I noticed that unlike with the nuScenes dataset where the forward direction (vehicle X axis) of model predictions and the ground truth were to the right, for me they had to point to the bottom (90 deg CW rotation), so I was wondering why. My data collection setup is pretty similar to nuScenes,  one lidar, 6 cameras at the same angles and with the same resolution. The only difference is that unlike the nuScenes dataset where the lidar X direction points to the right, mine points forward (same as the vehicle X axis), so I wonder if that's been causing the issue.

baiguosummer, 2024-04-23:
Hi, I read in the link you shared about the work on how to generate simulated nuscenes datasets using carla and simbev, I'm very interested in this, may I ask why the previous link to the repository is gone?

baiguosummer, 2024-04-23:
Hi, I read in the link you shared about the work on how to generate simulated nuscenes datasets using carla and simbev, I'm very interested in this, may I ask why the previous link to the repository is gone? @GoodarzMehr

GoodarzMehr, 2024-04-23:
Hi,
SimBEV originally only supported CARLA's Town06 map, but I recently made it private to work on expanding it to all CARLA maps. I have a working prototype and hopefully I'll make it public in a month or so.

closed 2024-05-04 by zhijian-liu