mit-han-lab/bevfusion #569: KeyError: "NuScenesDataset: 'infos'"

hankeceli, 2023-12-14:
I am in the process of building a dataset using `v1.0-mini`.

I have installed all the prerequisite libraries as described in the README.md file. Following that, I executed the `python setup.py develop` command for code base installation.

My working tree currently looks like this:

```
bevfusion
├── mmdet3d
├── tools
├── configs
├── data
│   ├── nuscenes
│   │   ├── maps
│   │   ├── samples
│   │   ├── sweeps
│   │   ├── v1.0-mini
│   │   ├── nuscenes_infos_train.pkl
│   │   ├── nuscenes_infos_val.pkl
│   │   ├── nuscenes_infos_mini_val.pkl
│   │   ├── nuscenes_infos_mini_train.pkl

```
I downloaded the .pkl files from this [link](https://github.com/open-mmlab/mmdetection3d/issues/2454#issuecomment-1517743436).
When I run the command `python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --version v1.0-mini`, I encounter a `FileNotFoundError: [Errno 2] No such file or directory: './data/nuscenes/nuscenes_infos_train.pkl'` error, unless I manually add the .pkl files shared by the author. As far as I understand, those .pkl files should automatically created, but it gives not found error during running Dataset Preparation phase [as it says here](https://github.com/open-mmlab/mmdetection3d/blob/1.0/docs/en/datasets/nuscenes_det.md#dataset-preparation).

If I add the .pkl files, I face the following error:
```
======
Loading NuScenes tables for version v1.0-mini...
23 category,
8 attribute,
4 visibility,
911 instance,
12 sensor,
120 calibrated_sensor,
31206 ego_pose,
8 log,
10 scene,
404 sample,
31206 sample_data,
18538 sample_annotation,
4 map,
Done loading in 0.208 seconds.
======
Reverse indexing ...
Done reverse indexing in 0.0 seconds.
======
total scene num: 10
exist scene num: 10
train scene: 8, val scene: 2
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] 404/404, 5.9 task/s, elapsed: 69s, ETA:     0s
nuscenes
train sample: 323, val sample: 81
Create GT Database of NuScenesDataset
Traceback (most recent call last):
  File "/home/ubuntu/Documents/students/han_k/hanBev/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/ubuntu/Documents/students/han_k/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/home/ubuntu/Documents/students/han_k/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/home/ubuntu/Documents/students/han_k/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 203, in load_annotations
    data_infos = list(sorted(data["infos"], key=lambda e: e["timestamp"]))
KeyError: 'infos'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 122, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/home/ubuntu/Documents/students/han_k/bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/home/ubuntu/Documents/students/han_k/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/home/ubuntu/Documents/students/han_k/hanBev/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
KeyError: "NuScenesDataset: 'infos'"
```

I also downloaded the map extension, but there are four versions of [them](https://www.nuscenes.org/download), and two of them are deprecated. Which one should be downloaded? Additionally, the directory structure for the map extensions is not described in the README.md. Should we directly extract all the files under the `maps` dir?

Thank you.

PS: It is crucial to run it on the NuScenes mini dataset. Looking forward for your solution.

closed 2023-12-21 by hankeceli

EpicGilgamesh, 2023-12-30:
Hi, how you managed to solve this issue? I'm using full-size Nuscenes dataset

hankeceli, 2023-12-30:
> Hi, how you managed to solve this issue? I'm using full-size Nuscenes dataset

Hello, I managed to solve the issue by following [this](https://github.com/mit-han-lab/bevfusion/issues/480#issuecomment-1667112241) method.

EpicGilgamesh, 2023-12-31:
Thank you very much I will try it!
Have you been preparing data using create_data.py or the command from the mmdet repo? 
python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes
https://github.com/open-mmlab/mmdetection3d/blob/1.0/docs/en/datasets/nuscenes_det.md

hankeceli, 2023-12-31:
> Thank you very much I will try it! Have you been preparing data using create_data.py or the command from the mmdet repo? python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir [...]

I followed in my previously mentioned [instruction](https://github.com/mit-han-lab/bevfusion/issues/569#issuecomment-1872588978), then I run the command as `python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --version v1.0-mini`

Dataset preparation of BEVFusion reference to the [mmdetection3d](https://github.com/open-mmlab/mmdetection3d/blob/1.0/docs/en/datasets/nuscenes_det.md). So, I just referenced to it for dataset preparation, I followed BEVFusion instructions for the rest of the steps.