mit-han-lab/bevfusion #570: Using a single image as input and limiting the field of view of the point cloud (nuScenes)

J4nekT, 2023-12-17:
First of all, I would like to thank you for publishing the code and for the great work. I don't have much experience with machine learning so I appreciate any help.
To adapt the pipeline to my system, I would like to make some changes to the input.

Firstly, I would like to use only one image as input instead of six. I would still like to use the nuscene data set, but only the front camera as input. What adjustments do I need to make and where?

Secondly, I would like to limit the point cloud to a FOV of 70 degrees. Do I have to adjust the point_cloud_range and/or the voxel_size? If further adjustments are necessary, please let me know.

Thank you in advance for your comments and suggestions.

BeMuCa, 2023-12-18:
I am also stuck trying to change configs so that I can do inference without lidar data (lidar failure), but I can give you a little input for the 2. question:

I came across this from the transfusion paper: " We set the region-ofinterest (RoI) to [−54.0m, 54.0m] for X and Y axis, and
[−5.0m, 3.0m] for Z axis" Now similiar to this in [configs/nuscenes](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml) we have: "point_cloud_range: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0]" compared with the structure seen in the configs of Transfusion you see that for us its the same for Z, but 51.2 for x and Y.
Then you can calculate your 70 degrees and adjust the range.

Hope that helps a little

EpicGilgamesh, 2023-12-22:
Hi! I see that your issue is kinda new. When trying to train the nuscenes dataset you encountered an error "no specified protocol"?

BeMuCa, 2023-12-22:
> Hi! I see that your issue is kinda new. When trying to train the nuscenes dataset you encountered an error "no specified protocol"?

It appears while doing inference too, but thats not an error, just a info

EpicGilgamesh, 2023-12-22:
Hm i get it. Could you tell me did you do anything that is not in the instruction to make this work?
Currently I'm stuck with some openmpi error

Poodlee, 2024-01-20:
Hello. I also wanted to use one image and about 60 degrees FOV this time, so I found it while looking for it. Could you please share what you have done about it?

gerardmartin2, 2024-04-08:
@J4nekT Have you been able to do it?

closed 2024-07-31 by zhijian-liu