mit-han-lab/bevfusion #578: KeyError: 'location'  while test seg

Qizhi697, 2023-12-30:
RUN `torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map`
and got error
```sh
load checkpoint from local path: pretrained/bevfusion-seg.pth
[                                                  ] 0/6019, elapsed: 0s, ETA:Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/opt/conda/lib/python3.8/site-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
    for i, data in enumerate(data_loader):
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/opt/conda/lib/python3.8/site-packages/torch/_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/bevfusion/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
    return self.prepare_test_data(idx)
  File "/root/bevfusion/mmdet3d/datasets/custom_3d.py", line 182, in prepare_test_data
    example = self.pipeline(input_dict)
  File "/opt/conda/lib/python3.8/site-packages/mmdet/datasets/pipelines/compose.py", line 41, in __call__
    data = t(data)
  File "/root/bevfusion/mmdet3d/datasets/pipelines/loading.py", line 294, in __call__
    location = data["location"]
KeyError: 'location'

--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
```
The command for detection test sucess, but seg test failed.
I use docker env and generate the .pkl files with bevfusion/tools/create_data.py.

can you help me with this? thanks!

971022jing, 2024-01-05:
> RUN `torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml pretrained/bevfusion-seg.pth --eval map` and got error
> 
> ```shell
> load checkpoint from local path: pretrained/bevfusion-seg.pth
> [                                                  ] 0/6019, elapsed: 0s, ETA:Traceback (most recent call last):
>   File "tools/test.py", line 230, in <module>
>     main()
>   File "tools/test.py", line 203, in main
>     outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
>   File "/opt/conda/lib/python3.8/site-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
>     for i, data in enumerate(data_loader):
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
>     data = self._next_data()
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
>     return self._process_data(data)
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
>     data.reraise()
>   File "/opt/conda/lib/python3.8/site-packages/torch/_utils.py", line 434, in reraise
>     raise exception
> KeyError: Caught KeyError in DataLoader worker process 0.
> Original Traceback (most recent call last):
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
>     data = fetcher.fetch(index)
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
>     data = [self.dataset[idx] for idx in possibly_batched_index]
>   File "/opt/conda/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
>     data = [self.dataset[idx] for idx in possibly_batched_index]
>   File "/root/bevfusion/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
>     return self.prepare_test_data(idx)
>   File "/root/bevfusion/mmdet3d/datasets/custom_3d.py", line 182, in prepare_test_data
>     example = self.pipeline(input_dict)
>   File "/opt/conda/lib/python3.8/site-packages/mmdet/datasets/pipelines/compose.py", line 41, in __call__
>     data = t(data)
>   File "/root/bevfusion/mmdet3d/datasets/pipelines/loading.py", line 294, in __call__
>     location = data["location"]
> KeyError: 'location'
> 
> --------------------------------------------------------------------------
> Primary job  terminated normally, but 1 process returned
> a non-zero exit code. Per user-direction, the job has been aborted.
> ```
> 
> The command for detection test is successful, but seg test failed. I use docker env and generate the .pkl files with bevfusion/tools/create_data.py.
> 
> can you help me with this? thanks!
hello，have you solved the problem?

Qizhi697, 2024-01-05:
No, I haven't found a solution yet.

EpicGilgamesh, 2024-01-16:
I think I found solution on this chinese site XD 
https://zhuanlan.zhihu.com/p/675643639

EpicGilgamesh, 2024-01-16:
Nope, doesn't work

zhijian-liu, 2024-05-04:
Please refer to https://github.com/mit-han-lab/bevfusion/issues/514.

closed 2024-05-04 by zhijian-liu