mit-han-lab/bevfusion #579: Error when training KeyError: "NuScenesDataset: 'infos'"

EpicGilgamesh, 2023-12-30:
Hello,
I'm getting this error while training using nuscenes dataset and torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth
Error: 

Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 203, in load_annotations
    data_infos = list(sorted(data["infos"], key=lambda e: e["timestamp"]))
KeyError: 'infos'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/train.py", line 87, in <module>
    main()
  File "tools/train.py", line 66, in main
    datasets = [build_dataset(cfg.data.train)]
  File "/bevfusion/mmdet3d/datasets/builder.py", line 38, in build_dataset
    dataset = CBGSDataset(build_dataset(cfg["dataset"], default_args))
  File "/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
KeyError: "NuScenesDataset: 'infos'"

liuyuansheng624, 2024-01-11:
> Hello, I'm getting this error while training using nuscenes dataset and torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth Error:
> 
> Traceback (most recent call last): File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg return obj_cls(**args) File "/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in **init** super().**init**( File "/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in **init** self.data_infos = self.load_annotations(self.ann_file) File "/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 203, in load_annotations data_infos = list(sorted(data["infos"], key=lambda e: e["timestamp"])) KeyError: 'infos'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "tools/train.py", line 87, in main() File "tools/train.py", line 66, in main datasets = [build_dataset(cfg.data.train)] File "/bevfusion/mmdet3d/datasets/builder.py", line 38, in build_dataset dataset = CBGSDataset(build_dataset(cfg["dataset"], default_args)) File "/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset dataset = build_from_cfg(cfg, DATASETS, default_args) File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg raise type(e)(f'{obj_cls.**name**}: {e}') KeyError: "NuScenesDataset: 'infos'"

请问你解决这个问题了吗

sunpian1, 2024-05-10:
use nuscenes_converter.py

ydj520, 2024-05-11:
> use nuscenes_converter.py

how to do it?

sunpian1, 2024-05-16:
find the file nuscenes_converter.py,you will know