mit-han-lab/bevfusion #580: Key Error: 'ann_info' when evaluating on Nuscenes test set

vtghsr, 2023-12-31:
I know the test set of Nuscenes does not have annotation, but how do I solve this problem?
[image]

alexzhuuuu, 2024-01-20:
> I know the test set of Nuscenes does not have annotation, but how do I solve this problem? [...]
change the codes in NuScenesDataset get_data_info function:
if not self.test_mode:                      
        annos = self.get_ann_info(index)        
        data['ann_info'] = annos          
        
return data

Maybe it works.

esh64, 2024-02-07:
Same problem, followed the #233 but got this same error. Does anybody know how to solve this?

closed 2024-05-04 by zhijian-liu