mit-han-lab/bevfusion #582: Performance is worse while training

186big, 2024-01-03:
Thank u for your great job.
I use your code and pretrained model to train a Lidar + Camera detection model in Nuscene dataset , but the more epochs I trained, the worse it  got. 
The loss is lower But the map is lower too.
[image]

971022jing, 2024-01-05:
> Thank u for your great job. I use your code and pretrained model to train a Lidar + Camera detection model in Nuscene dataset , but the more epochs I trained, the worse it got. The loss is lower But [...]
hello，have you solved the problem?

186big, 2024-01-05:
> Thank u for your great job. I use your code and pretrained model to train a Lidar + Camera detection model in Nuscene dataset , but the more epochs I trained, the worse it got. The loss is lower But [...]

sorry, not yet

Li-Whasaka, 2024-01-23:
using full dataset?

186big, 2024-01-23:
> using full dataset?

Yes

Acc2386, 2024-06-08:
same problem, have you salved this?

closed 2024-07-31 by zhijian-liu