mit-han-lab/bevfusion #587: Image BEV feature was getting worse during training

sangjinpark97, 2024-01-11:
Hi

I'm trying to train the BEVFusion using a single image and its corresponding point cloud for my custom dataset.

But, the performance gap was not large between LiDAR and fusion model.

So, I visualized the image features at a specific iteration during training. I observed that image bev feature was getting worse. It means that It cannot represent the scene in BEV map. I think the depth estimation seems to be a problem.

iteration 0.
[image]

iteration 1900
[image]

I think the image bev feature is enforced to be zero map during training because it didn't provide useful information for object detection.

Can you give me some advice?
Thank you in advance.

gerardmartin2, 2024-04-08:
Hello @eugenebak, do you remeber the changes you have done to the code?

zhijian-liu, 2024-05-04:
Adding support for custom dataset is beyond the scope of this codebase, and unfortunately, we don't have the bandwidth with such customized requests.

closed 2024-05-04 by zhijian-liu