mit-han-lab/bevfusion #589: Pre-processing Error  NuScenesDataset: [Errno 2] No such file or directory: 'nuscenes_infos_train.pkl'

anithselva, 2024-01-15:
Hi all,
I'm having trouble preprocessing the nuscenes dataset using the instructions on the repo

To reproduce:
`python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes`

I tried using an existing nuscenes_infos_train.pkl I found in one existing issue, but it gave me some downstream error when trying to run evaluation of bevfusion. I would prefer to be able to fix this at this level. 

Has anyone else rectified this?

```
Create GT Database of NuScenesDataset
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 52, in build_from_cfg
    return obj_cls(**args)
  File "/home/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 143, in __init__
    super().__init__(
  File "/home/bevfusion/mmdet3d/datasets/custom_3d.py", line 66, in __init__
    self.data_infos = self.load_annotations(self.ann_file)
  File "/home/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 202, in load_annotations
    data = mmcv.load(ann_file)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/io.py", line 60, in load
    with BytesIO(file_client.get(file)) as f:
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 992, in get
    return self.client.get(filepath)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/fileio/file_client.py", line 517, in get
    with open(filepath, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/dataset/nuscenes/nuscenes_infos_train.pkl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/create_data.py", line 101, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 45, in nuscenes_data_prep
    create_groundtruth_database(
  File "/home/bevfusion/tools/data_converter/create_gt_database.py", line 252, in create_groundtruth_database
    dataset = build_dataset(dataset_cfg)
  File "/home/bevfusion/mmdet3d/datasets/builder.py", line 42, in build_dataset
    dataset = build_from_cfg(cfg, DATASETS, default_args)
  File "/opt/conda/lib/python3.8/site-packages/mmcv/utils/registry.py", line 55, in build_from_cfg
    raise type(e)(f'{obj_cls.__name__}: {e}')
FileNotFoundError: NuScenesDataset: [Errno 2] No such file or directory: '/dataset/nuscenes/nuscenes_infos_train.pkl'
```

EpicGilgamesh, 2024-01-16:
Copy .pkl files from nuscenes folder in workspace to your data/nuscenes

EpicGilgamesh, 2024-01-16:
when you are in your workspace- the one with configs, tools, mmdet, etc. try going into nuscenes folder( no the one in the data, but the one in workspace)

anithselva, 2024-01-17:
> when you are in your workspace- the one with configs, tools, mmdet, etc. try going into nuscenes folder( no the one in the data, but the one in workspace). Are there any .pkl inside?

There is two pkl files that were generated, but they are not the ones that is part of the error:
nuscenes_infos_train_radar.pkl  
nuscenes_infos_val_radar.pkl

Is it looking for these files, but just not named correctly and placed in the correct folder?

EpicGilgamesh, 2024-01-17:
inside of that folder should be 2 more .pkls- the ones mentioned in your errors. They should be then copied to /dataset/nuscenes/nuscenes_infos_train.pkl. I think I saw somebody upload them at one of the issues

EpicGilgamesh, 2024-01-17:
Also look for solution in issue #480

anithselva, 2024-01-18:
> Also look for solution in issue #480

@EpicGilgamesh Ah! I don't know how I missed this. I was searching for `nuscenes_infos_train` but this issue didn't come up. 
Thank you for sharing. Will try this approach and will follow up.

anithselva, 2024-01-18:
@EpicGilgamesh 

Was able to resolve the issue. The link to https://github.com/mit-han-lab/bevfusion/issues/480 helped. I also fixed a typo in the info_path. Sharing my code changes to help others who may run into this in the future:

Specifically in  [tools/data_converter/nuscenes_converter.py](https://github.com/mit-han-lab/bevfusion/blob/main/tools/data_converter/nuscenes_converter.py#L95-L101), not only renaming the pkl filenames to remove `radar`, but also fixing the info_path/info_val_path to use `root_dir` instead of `info_prefix` to be consistent with the structure documented in https://mmdetection3d.readthedocs.io/en/latest/advanced_guides/datasets/nuscenes.html

from:
```        
info_path = osp.join(info_prefix, '{}_infos_train_radar.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(info_prefix, '{}_infos_val_radar.pkl'.format(info_prefix))
mmcv.dump(data, info_val_path)
```

to:
```
info_path = osp.join(root_path, '{}_infos_train.pkl'.format(info_prefix))
mmcv.dump(data, info_path)
data['infos'] = val_nusc_infos
info_val_path = osp.join(root_path, '{}_infos_val.pkl'.format(info_prefix))
mmcv.dump(data, info_val_path)
```

Thank you for your help!