mit-han-lab/bevfusion #590: learning rate configuration

konyul, 2024-01-19:
Thank you for sharing such a great work!
I am struggling to reproduce the L+C result of Bevfusion.
Currently, the repreduced result is 70.14 NDS.
What I did was modifying sweeps from 0 -> 9.
And I am now trying to freeze the img_backbone pipeline now.

If the above work doesn't work, then I am going to do following thing.
1. Modifying lr schedule
I reproduced the bevfusion in mmdetection3d. So I am comparing between the bevfusion in mmdetection3d and bevfusion in this repository.
The bevfusion in mmdetection3d and bevfusion in this repository both uses CosineAnnealingLR config but is slightly different.
lr used in bevfusion in mmdetection3d
[image]
the lr seems to decrease during the epoch
however lr used in bevfusion in this repository is 
[image]
It seems static during the epoch
So is there any way to change the lr config in mmdetection3d to bevfusion in this repository.

Is there any other way to try to reproduce?

Li-Whasaka, 2024-01-23:
have you reproduced the results?

gerardmartin2, 2024-04-08:
@konyul Any updates?

closed 2024-07-31 by zhijian-liu