mit-han-lab/bevfusion #594: CAN'T REPRODUCE  THE  RESULT IN PAPER

Li-Whasaka, 2024-01-26:
I ported bevfusion to my own project, but the trained mAP is very low. I don't know what the situation is. I have tried to adjust and modify many times, can anyone help me find some reasons for using it？
Here are my configuration files and log files：
[20240125_192959.log](https://github.com/mit-han-lab/bevfusion/files/14063180/20240125_192959.log)

```
_base_ = [#'../_base_/datasets/nus-3d.py',
          '../_base_/default_runtime.py']

voxel_size = [0.075, 0.075, 0.2]
point_cloud_range = [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]
class_names = [
    'car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier',
    'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'
]
metainfo = dict(classes=class_names)
input_modality = dict(use_lidar=True, use_camera=True)
backend_args = None
data_config={
    'cams': ['CAM_FRONT_LEFT', 'CAM_FRONT', 'CAM_FRONT_RIGHT',
             'CAM_BACK_LEFT', 'CAM_BACK', 'CAM_BACK_RIGHT'],
    'Ncams': 6,
    'input_size': (256, 704),
    'src_size': (900, 1600),

    # Augmentation
    'resize': (-0.06, 0.11),
    'rot': (-5.4, 5.4),
    'flip': True,
    'crop_h': (0.0, 0.0),
    'resize_test':0.04,
}

device='cuda'
model = dict(
    type = 'BEVFusion',
    encoders = dict(
        camera = dict(
            backbone = dict(
                type = 'SwinTransformer',
                embed_dims = 96,
                depths = [2, 2, 6, 2],
                num_heads = [3, 6, 12, 24],
                window_size = 7,
                mlp_ratio = 4,
                qkv_bias = True,
                qk_scale = None,
                drop_rate = 0.0,
                attn_drop_rate = 0.0,
                drop_path_rate = 0.2,  
                patch_norm = True,
                out_indices = [1, 2, 3],
                with_cp = False,
                convert_weights = True,
                init_cfg=dict(
                    type='Pretrained',
                    checkpoint= 'pretrained/checkpoint/swint-nuimages-pretrained.pth' # noqa: E251
                    # 'https://github.com/SwinTransformer/storage/releases/download/v1.0.0/swin_tiny_patch4_window7_224.pth'  # noqa: E501
                    )

                # pretrained='torchvision://resnet50',
                # # pretrained='/home/lwx/Desktop/lwx/distill-bev/resnet50.pth',
                # type='ResNet',
                # depth=50,
                # num_stages=4,
                # out_indices=(1,2, 3),
                # frozen_stages=-1,
                # norm_cfg=dict(type='BN', requires_grad=True),
                # norm_eval=False,
                # with_cp=True,
                # style='pytorch'
[... 326 lines cut ...]
        # map_classes=None,
        modality=input_modality,
        test_mode=False,
        use_valid_flag=True,
        box_type_3d='LiDAR',
    )
)

# learning rate
lr = 0.0001
# Optimizer
optimizer = dict(type='AdamW', lr=2e-4, weight_decay=0.01)
optimizer_config = dict(
    grad_clip=dict(max_norm=35,norm_type=2)
)
lr_config = dict(
    policy='step',
    warmup='linear',
    warmup_iters=500,
    warmup_ratio=0.001,
    step=[16, 22])
runner = dict(type='EpochBasedRunner', max_epochs=24)
evaluation = dict(interval=1)
checkpoint=dict( interval=1)
```

gerardmartin2, 2024-04-13:
@Li-Whasaka  Have you been able to solve it ?

Li-Whasaka, 2024-05-06:
> @Li-Whasaka Have you been able to solve it ?

Not resolved, it may be different from the underlying library because the module library is not part of this repository

closed 2024-05-26 by Li-Whasaka