mit-han-lab/bevfusion #607: The problem with the test

Ange1ika, 2024-04-15:
The problem with the test

[image]

if run
torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
he falls out somewhere, into an endless cycle - it is unclear what he is doing...
[image]

I tried to do debugging, at the last it outputs 185 and loads checkpoints

zhijian-liu, 2024-05-04:
Could you try `torchpack dirt-run -v -np 8 ...` to see if there's a more detailed error message?

Ange1ika, 2024-05-07:
I'm sorry that I took a long time to answer, now I will answer every day, because there is an urgent need to complete the project. There is such an error hanging now:

[image]
Thanks for the answer

Ange1ika, 2024-05-07:
[image]

Ange1ika, 2024-05-07:
torchpack dist-run -np 8 python visualize.py /configs/nuscenes/seg/fusion-bev256d2-lss.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/bevfusion-seg.pth

torchpack dist-run -np 1 python tools/visualize.py configs/nuscenes/seg/lidar-centerpoint-bev128.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/lidar-only-seg.pth
Can you tell me what the errors are? 

torchpack dist-run -np 2 python tools/visualize.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml --model pred --checkpoint pretrained/bevfusion-seg.pth --out-dir result/visualize fixed it, but
[create_data question](https://github.com/mit-han-lab/bevfusion/issues/461#top)
#461   
[image]

Ange1ika, 2024-05-07:
And can you explain how to figure out what is the dimension of the input data, for example, for the lidar-only-det(seg) model, what are the outputs and their dimension ? Is there any way to withdraw it? Sorry for the number of questions

Ange1ika, 2024-05-08:
https://github.com/mit-han-lab/bevfusion/issues/512   After installation, there was still an error... I started it 
torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml

Ange1ika, 2024-05-08:
[image]
 back to the last message

Ange1ika, 2024-05-15:
I've solved almost all those problems. 
[image]
There was a problem during training - [RuntimeError: sigmoid_focal_loss_forward_impl: implementation for device cuda:0 not found. · Issue #228 · mit-han-lab/bevfusion (github.com)](https://github.com/mit-han-lab/bevfusion/issues/228) ---I have cuda 12.3, and docker 11.3 - most likely, this is the problem, but the solutions presented here did not help me.

Ange1ika, 2024-05-15:
I can ask you to describe the dimensions of the main modules of the model for understanding. The article contains fragments, but it is difficult to put them together

closed 2024-07-31 by zhijian-liu