mit-han-lab/bevfusion #611: what is the lidar-only-det.pth

wyf0414, 2024-04-24:
This is the training code:
`torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml --model.encoders.camera.backbone.init_cfg.checkpoint pretrained/swint-nuimages-pretrained.pth --load_from pretrained/**lidar-only-det.pth**`

I want to know what is the **lidar-only-det.pth**. It's been used as a lidar-pretrained model.

I tried to val the **lidar-only-det.pth**, but got 0 mAP. Why?

Ask for your help.

zhijian-liu, 2024-05-03:
The configuration file you used is for the fusion model. Please use the configuration for LiDAR-only detection instead: https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/det/transfusion/secfpn/lidar/voxelnet_0p075.yaml.

closed 2024-05-04 by zhijian-liu

wyf0414, 2024-05-05:
Thanks very much!