mit-han-lab/bevfusion #612: Reproducing code with low accuracy

xmutyjs, 2024-04-25:
Hello, I am reproducing your code, but my training accuracy is far from reaching. How should I handle this? The training is for bevfusion det(L+C), which can output two modalities through visualization, but the training accuracy is too low(NDS=0.4665). We earnestly request your reply and help!

xmutyjs, 2024-04-25:
The system is ubuntu20.04, and the GPU is a single a100. Although there are four a100, parallel training is not feasible and can only temporarily use a single GPU

wyf0414, 2024-04-30:
Parallel training may affect performance. 
Try adding `CUDA_VISIBLE-DEVICES=0, 1, 2, 3` before the training code.

Another reason may be that The model does not converge.
Try adding the epoch.

PS: I'm a beginner. My opinion is for reference only.

closed 2024-05-04 by zhijian-liu