mit-han-lab/bevfusion #619: The difference between train and test AND what is the training strategy

Li-Whasaka, 2024-05-26:
When I try to train bevfusion to get mAP68.5,  it failed and just up to mAP62.8（trained epoch 22）, But I can get the right mAP68.5 by testing(test.py) bevfusion with bevfusion-det.pth provided by the official. 
SO, I'm curious about the train strategy. 

**device：**
    4090 x 4  
**Key-configs：**
    （1）samples_per_gpu=3
    （2）same model config as the official one
    （3）same train and test pipeline as the official one
    （4）same  LR config as the official one 
I don't know if the training results are related to LR and sample size OR anything else.

gerardmartin2, 2024-05-26:
Hi, are you using the version with radar (BEVFusion-R)? In my case something similar happened because I was trying to replicate the original BEVFusion (LIDAR+cam) using the new code version (that also adds radar data). I though that only changing in configs the parameter modality.use_radar to False was enough, but it is not the case. If this happens I think it is better to clone the previous version.

Li-Whasaka, 2024-05-27:
> Hi, are you using the version with radar (BEVFusion-R)? In my case something similar happened because I was trying to replicate the original BEVFusion (LIDAR+cam) using the new code version (that [...]

I use old version, It doesn't include radar，
And I have found the reason why I can't reproduce the result, It's related to the LR and sample size,  just increase the number of samples OR adjust LR as well as other learning strategy corresponding to sample size.

closed 2024-05-27 by Li-Whasaka

gerardmartin2, 2024-05-27:
I see, I am experiencing now some issues with that. How do you adapt the lr schedule? I use 1 sample per gpu and 3 gpus (so batch size is 3*1 = 3). Since the original batch size is 32 (8gpus * 4samples) I have decreased the initial lr proportionally by 1/10. However, the training slows down a lot from epoch 5 onwards and does not finish. Can I ask which lr schedule do you use (and which results does it give)?

Li-Whasaka, 2024-05-28:
> I see, I am experiencing now some issues with that. How do you adapt the lr schedule? I use 1 sample per gpu and 3 gpus (so batch size is 3*1 = 3). Since the original batch size is 32 (8gpus * [...]

You can cumulate grad by using GradientCumulative

gerardmartin2, 2024-05-28:
Thanks a lot! Just to make it clear. I should use the same lr as the original and just add the gradient cumulative right? Which should be the value of 'cumulative iters' then? I do not get if this value is related with the global batch size or only with the samples per gpu.

Thanks in advance

Li-Whasaka, 2024-05-28:
> Thanks a lot! Just to make it clear. I should use the same lr as the original and just add the gradient cumulative right? Which should be the value of 'cumulative iters' then? I do not get if this value is related with the global batch size or only with the samples per gpu.
> 
> Thanks in advance

YES, you should use same lr,  you can set sample size as large as possible and I set cumulative-iters to 8.

gerardmartin2, 2024-05-29:
@Li-Whasaka Thanks again! I will try it soon.